tbnobody/OpenDTU #2678: OpenDTU floods Home Assistant Log with MQTT messages "HM-xxx RX Success" and "HM-xxx TX Requests"

Gitsarry, 2025-05-15:
### What happened?

Every 5 s I get two MQTT messages from OpenDTU from these into the Home Assistant Log (see screenshot):
HM-800 RX Success
HM-800 TX Requests
HM-800 TX Re-Request Fragment

[image]

This makes the Home Assistant Log Viewer uncomfortable to use, when I just want to have a quick glance about relevant log entries e.g. of the last hour.

I could not find out how to make OpenDTU stop sending those messages. They are not of interest for me.
I would like only to be informed if there would be TX/RX errors.

I tried to set all loglevel in OpenDTU to "error" (see screenshot), but that did not help either.

[image]

### To Reproduce Bug

Configure OpenDTU to log via MQTT to your Home Assistant MQTT Broker (see screenshot).

[image]

### Expected Behavior

Like it was in older (e.g. one year ago) versions of OpenDTU, there should not be this massive amount of log messages, that seem not to indicate any problem.

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.5.10

### What firmware variant (PIO Environment)?

generic

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Gitsarry, 2025-05-15:
OK I managed to get rid of those log entries by going to:

Home Assistant->Setup->Devices->HM-800

and disable those three "diagnosis" sensors (see screenshot):

click on it -> click on the top gear icon -> switch "active" toggle to "off" -> click "update" at the bottom.

Not sure if this a valid solution or rather a workaround? Are having other users also to take this steps to prevent the log flooding, or do other users not have those mass amount of "diagnosis" sensor entries?

[image]

tbnobody, 2025-05-15:
> Like it was in older (e.g. one year ago) versions of OpenDTU, there should not be this massive amount of log messages, that seem not to indicate any problem.

This has nothing todo with the new logging. It is related to the fact that RF stats where introduced some when last year.

tbnobody, 2025-05-15:
> Not sure if this a valid solution or rather a workaround?

Its not a workaround. Its the way it should work. If you don't need a sensor, just don't activate it (or deactivate it). If it is activated you will see the changes in the log.

closed 2025-05-15 by tbnobody

Gitsarry, 2025-05-15:
Maybe you still want to have a look into it, why those RF stats sensors are in the Home Assistant log?
Normally, in the HA log there are only very few entries e.g. when the user manually switches the state of a device, or automatization events are triggered.

I mean, I have many sensors (e.g. temperature), and I do not get a log entry when the temperature changes?