tbnobody/OpenDTU #2680: Minor: German word in MPPT VE.Direct verbose log

Squint5471, 2025-05-15:
### What happened?

When using a Victron MPPT via VE.Direct there is a german word "Frame kaputt" in the verbose logging (see log example).

### To Reproduce Bug

-     Have broken frames in your VE.Direct communication
-     activate MPPT VE.Direct verbose logging
-     see console log (e.g., 11:14:14.231 > Frame kaputt)

### Expected Behavior

Output should be changed: "Frame kaputt" -> "Broken frame".

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

2025.04.17

### What firmware variant (PIO Environment)?

generic_esp32s3_usb

### Relevant log/trace output

```Shell
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'PID' = '0XA058'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'FW' = '164'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'SER' = 'HQ23486MH7H'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'V' = '27580'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'I' = '34900'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'VPV' = '92570'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'PPV' = '986'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'CS' = '3'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'MPPT' = '1'
11:14:13.852 > [VE.Direct MPPT 10/9] Text Data 'OR' = '0X00000000'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'ERR' = '0'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'LOAD' = 'OFF'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H19' = '53766'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H20' = '189'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H21' = '989'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H22' = '495'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'H23' = '1012'
11:14:14.231 > [VE.Direct MPPT 10/9] Text Data 'HSDS' = '189'
11:14:14.231 > Frame kaputt
```

### Anything else?

See also (closed): https://github.com/hoylabs/OpenDTU-OnBattery/issues/1911

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-05-15:
> Output should be changed: "Frame kaputt" -> "Broken frame".

No it shouldn't :)
It is a small easteregg for me to have something to laugh at

ms49434, 2025-05-15:
fyi, kaput is a very common word in the UK already. https://www.merriam-webster.com/dictionary/kaput

tbnobody, 2025-05-15:
Thats nice to know... Its in the code since the early beginnings. The phrase was introduced by @sprinterfreak when he wrote the first parser in python during long nights :-)

Squint5471, 2025-05-15:
Ok no worries, just wanted to let you know.