tbnobody/OpenDTU #2681: [ISSUE] Autodiscover doesn't work in HA

ErikSabel, 2025-05-17:
### What happened?

Well, i configrated openDTU. Enabled auto discovery for MQTT. On home assistant side MQTT was already installed for another device and mosquitto.

With explorer mqtt i found out that the formating from openDTU seems not to be valid for HA

Payload;
{
  "name": "Voltage",
  "stat_t": "homeassistant/138272223184/0/voltage",
  "uniq_id": "138272223184_ch0_Voltage",
  "unit_of_meas": "V",
  "dev": {
    "name": "Omvormer 1",
    "ids": "138272223184",
    "cu": "http://192.168.0.200",
    "mf": "OpenDTU",
    "mdl": "HMT-1800/2250-6T",
    "sw": "v24.4.24",
    "via_device": "OpenDTU-646E8C_6581900"
  },
  "exp_aft": 30,
  "dev_cla": "voltage",
  "stat_cla": "measurement"
}

Remark from my end;
Payload filed | Correct field in HA
"stat_t" | "state_topic"
"uniq_id" | "unique_id"
"unit_of_meas" | "unit_of_measurement"
"dev" | "device"
"dev_cla" | "device_class"
"stat_cla" | "state_class"

Not the expert at all, but i think the naming is wrong in openDTU
I didn't find a work around, eg. change in HA only for this one in renaming or whatever

### To Reproduce Bug

It's always present

### Expected Behavior

THat auto discovery finds automatic the devices

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v24.4.24

### What firmware variant (PIO Environment)?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-05-17:
It's working perfectly fine in a lot of installations here... Please provide a screenshot of your mqtt settings.

ErikSabel, 2025-05-17:
> It's working perfectly fine in a lot of installations here... Please provide a screenshot of your mqtt settings.
> 
> > Not the expert at all, but i think the naming is wrong in openDTU
> 
> Why do you think the naming is wrong? Looks good for me. (This are valid shortcuts)
> 
> You didn't even write whats not working at all.

These are the settings in opendtu;

[image]

So what is not working is the auto discovery on home assistant side of the opendtu.
There is communication, as i showed. However it is not picked up by mqtt. With some googling i found that stat_t should be state_topic for example and that could be the reason of that auto discovery doesnt work

tbnobody, 2025-05-17:
> These are the settings in opendtu;

Please provide a screenshot of all settings....

> With some googling i found that stat_t should be state_topic for example and that could be the reason of that auto discovery doesnt work

Thats just not right. First, as I said, it's working in many installation without problems, and second, this short names are documented on the home assistant page.
https://www.home-assistant.io/integrations/mqtt/#discovery-payload

[image]

ErikSabel, 2025-05-17:
This is the screenshot of all settings;

[image]

[image]

> Thats just not right

That's why I am saying, I am not an expert :) First thing that I found as a trigger what could be wrong, but not knowing what's going on exacly

THis is my HA version;
Installatiemethode
Home Assistant OS
Core
2025.5.2
Supervisor
2025.05.1
Operating System
15.2
Frontend
20250516.0

This is the mosquitto log;
2025-05-17 13:07:14: mosquitto version 2.0.21 terminating
time="2025-05-17T13:07:14+02:00" level=error msg="http request error: Post \"http://127.0.0.1:80/superuser\": read tcp 127.0.0.1:58294->127.0.0.1:80: read: connection reset by peer"
time="2025-05-17T13:07:14+02:00" level=error msg="http request error: Post \"http://127.0.0.1:80/acl\": dial tcp 127.0.0.1:80: connect: connection refused"
time="2025-05-17T13:07:14+02:00" level=error msg="Post \"http://127.0.0.1:80/superuser\": read tcp 127.0.0.1:58294->127.0.0.1:80: read: connection reset by peer"
time="2025-05-17T13:07:14+02:00" level=error msg="http request error: Post \"http://127.0.0.1:80/superuser\": dial tcp 127.0.0.1:80: connect: connection refused"
time="2025-05-17T13:07:14+02:00" level=error msg="http request error: Post \"http://127.0.0.1:80/acl\": dial tcp 127.0.0.1:80: connect: connection refused"
time="2025-05-17T13:07:14+02:00" level=error msg="Post \"http://127.0.0.1:80/superuser\": dial tcp 127.0.0.1:80: connect: connection refused"
2025-05-17 13:07:14: Saving in-memory database to /data//mosquitto.db.
[11:07:14] INFO: Service restart after closing
[11:07:14] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/mosquitto.sh
[13:07:22] INFO: Certificates found: SSL is available
cont-init: info: /etc/cont-init.d/mosquitto.sh exited 0
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun mosquitto (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
[13:07:23] INFO: Starting NGINX for authentication handling...
s6-rc: info: service legacy-services successfully started
[13:07:23] INFO: Starting mosquitto MQTT broker...
2025-05-17 13:07:23: Warning: Mosquitto should not be run as root/administrator.
2025-05-17 13:07:23: mosquitto version 2.0.21 starting
2025-05-17 13:07:23: Config loaded from /etc/mosquitto/mosquitto.conf.
2025-05-17 13:07:23: Loading plugin: /usr/share/mosquitto/go-auth.so
2025-05-17 13:07:23:  ├── Username/password checking enabled.
2025-05-17 13:07:23:  ├── TLS-PSK checking enabled.
2025-05-17 13:07:23:  └── Extended authentication not enabled.
2025-05-17 13:07:23: Opening ipv4 listen socket on port 1883.
2025-05-17 13:07:23: Opening ipv6 listen socket on port 1883.
2025-05-17 13:07:23: Opening websockets listen socket on port 1884.
2025-05-17 13:07:23: Opening ipv4 listen socket on port 8883.
2025-05-17 13:07:23: Opening ipv6 listen socket on port 8883.
2025-05-17 13:07:23: Opening websockets listen socket on port 8884.
2025-05-17 13:07:23: mosquitto version 2.0.21 running
2025-05-17 13:07:24: New connection from 172.30.32.1:48219 on port 1883.
2025-05-17 13:07:24: New connection from ::1:39346 on port 1883.
2025-05-17 13:07:24: New client connected from 172.30.32.1:48219 as 3mfh7eHpsSBkNq5koBMI1E (p2, c1, k60, u'Mosquitto').
2025-05-17 13:07:24: Client <unknown> disconnected due to protocol error.
2025-05-17 13:07:24: New connection from 192.168.0.200:57421 on port 1883.
2025-05-17 13:07:24: New client connected from 192.168.0.200:57421 as OpenDTU-6581900 (p2, c1, k15, u'opendtu').
[13:07:25] INFO: Successfully send discovery information to Home Assistant.
[13:07:26] INFO: Successfully send service information to the Supervisor.
2025-05-17 13:07:29: New connection from 172.30.32.1:58977 on port 1883.
2025-05-17 13:07:29: New client connected from 172.30.32.1:58977 as 7EGB3Cf4ELEqLkqLa0CPlr (p1, c1, k60, u'addons').
2025-05-17 13:08:59: Client 3mfh7eHpsSBkNq5koBMI1E disconnected.
2025-05-17 13:09:40: New connection from 172.30.32.1:36837 on port 1883.
error: received null username or password for unpwd check
2025-05-17 13:09:40: Client 4gyGo0eAxg2f7YW6bQGYLa disconnected, not authorised.
2025-05-17 13:09:56: New connection from 172.30.32.1:39329 on port 1883.
2025-05-17 13:09:56: New client connected from 172.30.32.1:39329 as 5Iwandi9soRLJW8gqDovlI (p2, c1, k60, u'opendtu').
2025-05-17 13:09:56: Client 5Iwandi9soRLJW8gqDovlI disconnected.
2025-05-17 13:09:56: New connection from 172.30.32.1:49237 on port 1883.
2025-05-17 13:09:56: New client connected from 172.30.32.1:49237 as 3Bo7pmfEtPUIEsSUFhodYn (p2, c1, k60, u'opendtu').
2025-05-17 13:10:52: New connection from 192.168.0.200:64702 on port 1883.
2025-05-17 13:10:52: Client OpenDTU-6581900 already connected, closing old connection.
2025-05-17 13:10:52: New client connected from 192.168.0.200:64702 as OpenDTU-6581900 (p2, c1, k15, u'opendtu').

Any other ideas, required log and so on are welcome

tbnobody, 2025-05-17:
The Topic should not begin with omeassistant/ as you maybe override some home assistant internal states....... just use something like `solar/`.

ErikSabel, 2025-05-17:
> The Topic should not begin with omeassistant/ as you maybe override some home assistant internal states....... just use something like `solar/`.

Ok, I have changed it to solar/

Messages are received under solar/

[image]

Mosquitto log;
2025-05-17 16:32:46: New connection from 172.30.32.1:49217 on port 1883.
2025-05-17 16:32:46: New client connected from 172.30.32.1:49217 as 2cj3IMJLFTUFtn2BGIlydP (p2, c1, k60, u'opendtu').
2025-05-17 16:32:56: New connection from 172.30.32.1:47689 on port 1883.
2025-05-17 16:32:56: New client connected from 172.30.32.1:47689 as 04EKIxXcuB0wjBE6gUSPsd (p1, c1, k60, u'addons').
2025-05-17 16:35:35: Client 04EKIxXcuB0wjBE6gUSPsd closed its connection.
2025-05-17 16:35:56: Client 2cj3IMJLFTUFtn2BGIlydP closed its connection.
s6-rc: info: service legacy-services: stopping
2025-05-17 16:36:12: mosquitto version 2.0.21 terminating
2025-05-17 16:36:12: Saving in-memory database to /data//mosquitto.db.
[14:36:13] INFO: Service restart after closing
[14:36:13] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/mosquitto.sh
[16:37:23] INFO: Certificates found: SSL is available
cont-init: info: /etc/cont-init.d/mosquitto.sh exited 0
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun mosquitto (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
[16:37:24] INFO: Starting NGINX for authentication handling...
s6-rc: info: service legacy-services successfully started
[16:37:24] INFO: Starting mosquitto MQTT broker...
2025-05-17 16:37:24: Warning: Mosquitto should not be run as root/administrator.
2025-05-17 16:37:24: mosquitto version 2.0.21 starting
2025-05-17 16:37:24: Config loaded from /etc/mosquitto/mosquitto.conf.
2025-05-17 16:37:24: Loading plugin: /usr/share/mosquitto/go-auth.so
2025-05-17 16:37:24:  ├── Username/password checking enabled.
2025-05-17 16:37:24:  ├── TLS-PSK checking enabled.
2025-05-17 16:37:24:  └── Extended authentication not enabled.
2025-05-17 16:37:24: Opening ipv4 listen socket on port 1883.
2025-05-17 16:37:24: Opening ipv6 listen socket on port 1883.
2025-05-17 16:37:24: Opening websockets listen socket on port 1884.
2025-05-17 16:37:24: Opening ipv4 listen socket on port 8883.
2025-05-17 16:37:24: Opening ipv6 listen socket on port 8883.
2025-05-17 16:37:24: Opening websockets listen socket on port 8884.
2025-05-17 16:37:24: mosquitto version 2.0.21 running
2025-05-17 16:37:25: New connection from ::1:49042 on port 1883.
2025-05-17 16:37:25: Client <unknown> disconnected due to protocol error.
[16:37:26] INFO: Successfully send discovery information to Home Assistant.
2025-05-17 16:37:27: New connection from 192.168.0.200:55536 on port 1883.
[16:37:27] INFO: Successfully send service information to the Supervisor.
2025-05-17 16:37:27: New client connected from 192.168.0.200:55536 as OpenDTU-6581900 (p2, c1, k15, u'opendtu').
2025-05-17 16:38:30: New connection from 172.30.32.1:48571 on port 1883.
2025-05-17 16:38:30: New client connected from 172.30.32.1:48571 as 3Q2y3IMJpXK6jXJK7vLD5W (p2, c1, k60, u'opendtu').
2025-05-17 16:38:40: New connection from 172.30.32.1:42631 on port 1883.
2025-05-17 16:38:40: New client connected from 172.30.32.1:42631 as 09xu82574GGuVsrBAkXMSI (p1, c1, k60, u'addons').
2025-05-17 16:41:49: Client 3Q2y3IMJpXK6jXJK7vLD5W closed its connection.
2025-05-17 16:42:17: New connection from 172.30.32.1:33557 on port 1883.
2025-05-17 16:42:17: New client connected from 172.30.32.1:33557 as 4N7fU4x8yDoWuVQQAmDTed (p2, c1, k60, u'opendtu').
2025-05-17 16:44:39: Client 4N7fU4x8yDoWuVQQAmDTed closed its connection.
2025-05-17 16:45:10: New connection from 172.30.32.1:60351 on port 1883.
2025-05-17 16:45:10: New client connected from 172.30.32.1:60351 as 589ne7ZZme2vvIAsX5RQRs (p2, c1, k60, u'opendtu').
2025-05-17 16:45:46: New connection from 192.168.0.200:61972 on port 1883.
2025-05-17 16:45:46: Client OpenDTU-6581900 already connected, closing old connection.
2025-05-17 16:45:46: New client connected from 192.168.0.200:61972 as OpenDTU-6581900 (p2, c1, k15, u'opendtu').
2025-05-17 16:46:57: New connection from 172.30.33.4:42262 on port 1883.
2025-05-17 16:46:57: New client connected from 172.30.33.4:42262 as mqtt-explorer-52dfac40 (p2, c1, k60, u'Erik').

But still no auto detection of the openDTU device in MQTT. Also not present in the diagnostic file of mqtt

tbnobody, 2025-05-17:
Change the home assistant prefix to `homeassistant/`

ErikSabel, 2025-05-17:
> Change the home assistant prefix to `homeassistant/`

That was it, that solved it. The "/" behind homeassistant in the perfix. They are now visible!

stefan123t, 2025-05-18:
@tbnobody should we double check for a trailing slash `/` either in the VUE UI or API backend before allowing a new topic / prefix ?
Are there possible and valid scenarios where someone would want only a prefix without slash, e.g. OpenDTU1-topics, OpenDTU2-topics, etc. ?

closed 2025-07-02 by github-actions[bot]