tbnobody/OpenDTU #2688: yieldDay wird nicht zum Tageswechsel zurückgesetzt

ztjfjgkfdhrshrjzzfzujht, 2025-05-19:
### What happened?

I operate a battery storage system between a solar cell and an inverter (HM-1200-4T). The battery storage unit is currently configured so that it delivers a constant 170 watts from sunset. This means that openDTU also reports this value, which is correct. I have also connected openDTU to HomeAssistant using MQTT. On the topic solar/SERIALNUMBER/0/yieldday, however, the value is not reset to 0 at 0:00 on each day change. If the battery storage system delivers electricity until sunrise, the value is not reset at all. If the battery storage unit does not emit any more power before sunrise, the value is reset at sunrise.

### To Reproduce Bug

Firmware: v25.5.10

### Expected Behavior

reset at 0:00

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

3f01b8d

### What firmware variant (PIO Environment)?

generic_esp32

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

schlimmchen, 2025-05-19:
OpenDTU does not account for battery-powered inverters. Have a look at OpenDTU-OnBattery as an alternative, where an automatic restart of the inverter is implemented. Only a restart of the inverter can reset the yield value, there is no command to do so.

ztjfjgkfdhrshrjzzfzujht, 2025-05-19:
Thank you very much for your quick reply. I will have a look at openDTU-OnBattery.
And also many thanks that openDTU exists and that you invest your time!

closed 2025-05-19 by ztjfjgkfdhrshrjzzfzujht