tbnobody/OpenDTU #2730: "Total Yield Total" is reported as zero, when all inverters are offline (e.g. at night) and OpenDTU is restarted

Gitsarry, 2025-06-17:
### What happened?

I restarted OpenDTU at night, when all inverters where offline because lack of power, and after restart, OpenDTU reported `Total Yield Total` (MQTT entity `sensor.opendtu_xxxxxx_yield_total`) value of zero, instead of the last known (greater than zero) value it reported before the restart.

### To Reproduce Bug

- at day in normal operation, look at the OpenDTU reported `Total Yield Total` value. It is greater than zero.
- at night when all inverters are offline, because they have no power, look at the OpenDTU reported `Total Yield Total` value. It stays constantly at the last known value.
- at night when all inverters are offline, because they have no power, restart OpenDTU
- look at the OpenDTU reported `Total Yield Total` value. It reports zero.

### Expected Behavior

- The value of `Total Yield Total` (MQTT entity `sensor.opendtu_xxxxxx_yield_total`) is expected to be an ever only increasing value (like a miles counter of a car).
- The ever increasing total yield value should be stored persistently in OpenDTU
- If OpenDTU computes a total yield value that is less than the last known (maximum) value, it should report "unknown" and give a warning message in the GUI.
- OpenDTU could offer a configurable total yield offset value, to "correct" the reported value, e.g. if user bought a used OpenDTU, or removes an inverter

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

v25.5.10

### What firmware variant (PIO Environment)?

generic

### Relevant log/trace output

```Shell

```

### Anything else?

I monitor the value of yield total in Grafana (Home Assistant), and also compute a value of "yield in the last 24h" (difference of the current value and the value 24h ago).
It destroys the data if the yield total value temporarily gets zero.

In the past, I computed the total yield value myself in home assistant as sum of the inverter total yield values.

But some weeks ago, I saw that OpenDTU reports the sum already itself, so I thought you can make it simpler and just user the OpenDTU reported value.

So, workaround for me would be to switch back to the self computed sum value.

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

Gitsarry, 2025-06-17:
I have not tested it, but I _think/expect_ that the Home Assistant combinations(sum) sensor, which summed up the total yield values of my two inverters, returns "unknown" if not both inverters report a value (e.g. when at least one is offline due to lack of power, or for maintenance, or sold etc.). I would expect this also from OpenDTU: If not _all_ of the defined inverters report a total yield value, OpenDTU should report "unknown" as total yield value.

RomanDenes, 2025-06-25:
I don't know where the "Yield Total" is stored. I suspect HMS inverter. Every morning, when the inverter starts, it reports few wrong numbers (either "0" or "null"). This totally messes up Home assistant's statistics. Once it gets correct number, it adds the entire "Yield Total" as if it was consumed in that very moment.

I believe the openDTU shall implement some data integrity check and ignore all the "0" or "null" states (unless the inverter has been restarted recently).

closed 2025-08-30 by github-actions[bot]