tbnobody/OpenDTU #2742: [Request] Support of new Hoymiles Inverter-Series MIT-4000/4500/5000-8T

joe-ben-75, 2025-06-27:
### Is your feature request related to a problem? Please describe.

I'm currently using a Hoymiles HMT-2000 3-phase Inverter very successfully controlled by the "onBattery" fork of OpenDTU. With the intention to increase the Inverter power I thought about adding another HMT-2000 to my setup or to replace the HMT-2000 with a new MIT-4000-8T. Although the new inverter also uses sub 1G communication, it is not yet supported by OpenDTU. But I would prefer the single Inverter solution because I expect that it can be better controlled than two Inverters.

### Describe the solution you'd like

Please add the support for the new Hoymiles Inverter-Series MIT-4000/4500/5000-8T.

### Additional context

If there is something I can do to support the implementation of the enhancement please let me know.
I could buy one inverter and could support communication recording (if I'm told how to do this).

Geoffn-Hub, 2025-08-27:
Serial Number 1520A02

[19:46:29.103] D (1861127) hoymiles: TX RealTimeRunData 865.00 MHz --> 15 A0 11 52 4A 80 10 24 12 80 0B 00 68 AF 44 74 00 00 00 00 00 00 00 00 80 E6 00
[19:46:29.103] D (1861221) hoymiles: RX 865.00 MHz --> 95 A0 11 52 4A 80 10 24 12 01 00 01 01 33 01 2C 00 02 00 01 00 06 00 05 00 00 85 | -52 dBm
[19:46:29.103] D (1861265) hoymiles: RX 865.00 MHz --> 95 A0 11 52 4A 80 10 24 12 02 24 88 00 00 0C E8 04 C7 02 C9 01 34 01 31 00 01 DC | -53 dBm
[19:46:29.103] D (1861315) hoymiles: RX 865.00 MHz --> 95 A0 11 52 4A 80 10 24 12 03 00 01 00 03 00 03 00 00 62 BC 00 00 6C 51 03 C7 BF | -52 dBm
[19:46:29.456] D (1861366) hoymiles: RX 865.00 MHz --> 95 A0 11 52 4A 80 10 24 12 04 04 21 00 0C 00 00 00 00 00 00 00 00 00 00 01 7B CD | -53 dBm
[19:46:29.456] D (1861406) hoymiles: RX 865.00 MHz --> 95 A0 11 52 4A 80 10 24 12 85 00 09 D6 2B EB | -53 dBm

stefan123t, 2025-08-27:
@tbnobody this looks like a complete five packet answer from 0x01 .. 0x85. Can you try to decode this in your Excel sheet ?

Can you add preliminary support for MIT models ?

There might be a typo in @Geoffn-Hub Serial ID it should be 1520 A0 11 52 4A if I am not mistaken.

tbnobody, 2025-08-28:
As a first step I used the same logic as for the HMS 4 channel inverters

[image]

Strange thing is, that Yield Total is always zero and I wasn't able to find 3 AC voltages (just found 2 values with ~230v, segment three idx 8+9 and idx 12+13)

tbnobody, 2025-08-28:
[image]

Yield total makes more sense now. (Even if its very different between the channels, but maybe @Geoffn-Hub can say something about it)

The AC part looks very zero....

Geoffn-Hub, 2025-08-29:
Ok - Let's try again...

Serial numbers
 - Serial number for the new inverter is 1520A02556xx
 - I have an HMS-2000-4T inverter 1164a01152xx
 
It looks like the debug capture is actually the return from the old inverter - I can't see any entries in the RF debug log that match the serial number of the new inverter - not even a fetch inverter entry.  Will the code try to query an inverter with a serial number format that it cannot identify?

stefan123t, 2025-08-29:
No the OpenDTU will not try to communicate with an Inverter which has an unknown Prefix, i.e. 1520

Please use a fake Serial ID, ie with Prefix 1164 because the model is known and has atcleastvfour channels. It may have problems understanding the slightly longer response but we should be able to understand it from the traces.

tbnobody, 2025-08-29:
Unfortunatly thats not a complete packet. You should look for a `hoymiles: TX RealTimeRunData` which marks the begin of a transaction and then for 
```
hoymiles: RX Period End
hoymiles: Success
```
which marks the end of a successful received packet.

In between you should see a lot of `hoymiles: RX 865.00 MHz --> 95 A0 25.....` which is the important part.

stefan123t, 2025-08-30:
@Geoffn-Hub can you format your response as /code or enclose it with triple backticks ``` so it becomes more readable ?

Basically we are looking for the responses 01..86 to your RealTimeRunData request.

@tbnobody I do see some of the packets but I have the hunch that the MIT may send the response much faster. Ie. the first RealTimeRunData response has packets 01 and 04, whereas later in the previous (already formatted log) there are packets 05 and 86. 

Then there is another sequence of responses for the AlarmData Commamd which goes from 02, 05, 08 to 0B and "completes" a little later with 8C.

As Thomas wrote there is not a complete transmission, do you have built your OpenDTU yourself or do you use an OpenDTU Fusion board with proper build quality ? How is the reception with your inverters ?

Geoffn-Hub, 2025-09-01:
Firmware update applied - release notes for MIT-5000 V01.02.01-01.01.12 are as follows:
In the installation scenario with no leakage current requirement, the following fixes have been added
1. Temperature protection modification
2. Fix for unresponsive arc fault clearing command after disabling AFCI function
3. New anti-backflow (ABF) function added [including related configuration distribution and storage]
4. Logic change for disabling low-light detection
5. Optimized grid connection detection logic to solve low power startup issues
6. MPPT mode modification: reference given after power limit exit to solve repeated entry/exit of boost overvoltage loop
7. MPPT efficiency optimization
8. Grid connection regulation version update

Geoffn-Hub, 2025-09-01:
Alliance delivered another Fusion board DTU this afternoon - this is the log from that one

[opendtu-log-2025-09-01_18-29-39.txt](https://github.com/user-attachments/files/22081893/opendtu-log-2025-09-01_18-29-39.txt)

Geoffn-Hub, 2025-09-09:
Thanks - have spent some more time on this after having been away for the weekend - the success rate of receiving a complete response is very low, so it takes a while to log a complete response.

I've attached the whole log file

[image]

`
[08:21:13.924] D (52484199) hoymiles: TX RealTimeRunData 865.00 MHz --> 15 A0 25 56 6B 80 10 24 12 80 0B 00 68 BF C7 59 00 00 00 00 00 00 00 00 F6 84 BB
[08:21:13.924] D (52484338) hoymiles: RX 865.00 MHz --> 95 A0 25 56 6B 80 10 24 12 02 00 1F 00 62 00 00 3D 83 00 05 02 89 00 37 01 69 9B | -70 dBm
[08:21:14.351] D (52484489) hoymiles: RX 865.00 MHz --> 95 A0 25 56 6B 80 10 24 12 05 00 00 02 F9 00 00 00 F4 00 0B 00 0D 00 0B 03 C2 4D | -69 dBm
[08:21:14.609] D (52484739) hoymiles: TX RequestFrame 865.00 MHz --> 15 A0 25 56 6B 80 10 24 12 86 8D
[08:21:14.609] D (52484868) hoymiles: TX RequestFrame 865.00 MHz --> 15 A0 25 56 6B 80 10 24 12 86 8D
[08:21:14.609] D (52484943) hoymiles: RX 865.00 MHz --> 95 A0 25 56 6B 80 10 24 12 86 00 DC 00 03 00 00 00 00 00 0C 00 01 00 00 5F 46 C6 | -70 dBm
`

[opendtu-log-2025-09-09_08-21-17.txt](https://github.com/user-attachments/files/22227637/opendtu-log-2025-09-09_08-21-17.txt)

Geoffn-Hub, 2025-09-19:
@stefan123t @tbnobody - anything else I can be doing to help?

stefan123t, 2025-10-31:
> <img alt="Image" width="2000" height="141" src="https://private-user-images.githubusercontent.com/1753222/483266534-353b971c-aae1-4ec0-81d7-ac9e79a5e92e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjE4OTY2NzcsIm5iZiI6MTc2MTg5NjM3NywicGF0aCI6Ii8xNzUzMjIyLzQ4MzI2NjUzNC0zNTNiOTcxYy1hYWUxLTRlYzAtODFkNy1hYzllNzlhNWU5MmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTAzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEwMzFUMDczOTM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGY2YmU4MGVmNzhmMTBmZGUxMTUyODk3NzM4M2ViMTAzMDNmOWUxMWM2MDc2NGQzNTNlZjRhYzFlODVjZTBkZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.qzCpWC7de6PYperlbhRCoTJuRc8b2y-29VHp9axTTHw">
> Yield total makes more sense now. (Even if its very different between the channels, but maybe [@Geoffn-Hub](https://github.com/Geoffn-Hub) can say something about it)
> 
> The AC part looks very zero....

@tbnobody do you have your Excel Sheet actually linked somewhere in the OpenDTU repository ?
I recall there might still be an ancient version in the mikrocontroller.net Thread afaik.

stefan123t, 2026-01-28:
@DanielR95 @lumapu I recall you accomplished some serious efforts regarding the CMT2300A transceiver code back then in the Ahoy DTU. 

Can you please take a look at this issue. I have the hunch that we may need to use a different receive routine in order to not miss two-thirds of the packets.

Geoffn-Hub, 2026-02-01:
Finally got some time to take a proper look at this and try to understand how the code works, and to figure out an approach for capturing the communications with the MIT inverter - feel free to pull me up if I have misunderstood

The bottleneck seems to be in the loop() function of HoymilesRadio_CMT.cpp. Here's how the receive path appears to work:

Current flow:

1. CMT2300A radio receives a packet → triggers GPIO3 interrupt → sets _packetReceived = true
2. loop() runs (called from Arduino main loop) and checks _packetReceived
3. If true: reads all available packets from the FIFO into _rxBuffer (ring buffer)
4. If false: pops ONE packet from _rxBuffer and processes it (CRC check, serial matching, routing to inverter)
The problem for MIT inverters:
The MIT sends 6 response fragments very rapidly (back-to-back). The CMT2300A has a 64-byte merged FIFO. Each Hoymiles packet is ~27 bytes. So the FIFO can hold about 2 packets before it overflows.

I think that the issue is step 4 — packets are only processed one at a time, and only when no new packet is being received. If the MIT sends packets faster than the loop processes them:

• Packets pile up in the 64-byte hardware FIFO
• FIFO overflows before loop() can drain it
• 2 out of every 3 packets are lost
The available() function also has a subtle issue — it checks multiple flags (PREAM_OK, SYNC_OK, CRC_OK, PKT_OK) which may cause it to read before the packet is fully received.

It seems that two key changes are needed:

1. Drain the FIFO faster — process all buffered packets in a single loop() iteration instead of one-at-a-time
2. Reduce time between reads — the current code does CRC checking, serial matching, and inverter routing between FIFO reads. For capture mode, skip all that and just dump raw packets.

I've created a branch to try and implement this and allow for a promiscuous capture mode: https://github.com/Geoffn-Hub/OpenDTU/tree/feature/capture-mode

What's implemented:

4 files changed, 110 insertions:

1. HoymilesRadio_CMT.h — Added _captureMode flag + public getter/setter
2. HoymilesRadio_CMT.cpp — Modified the reception loop to log every valid-CRC frame before the DTU ID filter when capture mode is on. Each captured frame shows:  • Frequency (MHz) and RSSI (dBm)
  • Source serial (inverter) and destination serial (DTU) extracted from the frame
  • Frame length and full raw hex dump
  • Classification: "foreign traffic" / "unknown inverter" / normal processing
  • CRC failures are also logged with raw data in capture mode

3. WebApi_dtu.h/.cpp — REST API to toggle:  • GET /api/dtu/capture → returns {"capture_mode": true/false}
  • POST /api/dtu/capture with {"capture_mode": true} → enables/disables

All output goes through the existing ESP_LOGI → Web Console pipeline — no new UI needed, just open the console in the browser.

I've also started to stub out some placeholders for the MIT inverter
• MIT_8CH class with serial prefix 0x1520, type name "MIT-5000-8T"
• 8 DC channels (CH0-CH7) in paired MPPT groups sharing voltage readings
• 3-phase AC output (same layout as HMT-6CH)
• ChannelNum_t extended with CH6/CH7 — all arrays automatically sized via CH_CNT
• Byte assignment table is clearly marked as placeholder

Geoffn-Hub, 2026-02-01:
Raw frames

```
0x01: 01 00 01 01 51 00 67 01 5D 00 01 39 F0 00 25 01 62
0x02: 02 00 66 01 70 00 00 FE 2A 00 23 02 D3 00 7E 03 91
0x03: 03 00 02 5E 0B 00 58 02 B5 00 6F 03 03 00 02 50 0E
0x04: 04 00 64 09 0D 09 1C 09 1F 0F AB 0F D2 0F C1 13 88
0x05: 05 00 00 08 CE 00 00 00 E1 00 22 00 21 00 1E 03 E8
0x86: 86 00 78 00 03 00 00 00 00 00 1A 63 40
```

```
Off Sz  Hex          Raw    Value         Field
--- --  ----------  -----  -----------   --------
 DC CH0 (MPPT1)
  2  2  01 51         337   33.7 V       UDC
  4  2  00 67         103    1.03 A       IDC
  6  2  01 5D         349   34.9 W       PDC
  8  4  00 01 39 F0 80368   80.368 kWh   YT
 12  2  00 25          37   37 Wh        YD

 DC CH1 (MPPT2)
 14  2  01 62         354   35.4 V       UDC
 16  2  00 66         102    1.02 A       IDC
 18  2  01 70         368   36.8 W       PDC
 20  4  00 00 FE 2A 65066   65.066 kWh   YT
 24  2  00 23          35   35 Wh        YD

 DC CH2 (MPPT3)
 26  2  02 D3         723   72.3 V       UDC
 28  2  00 7E         126    1.26 A       IDC
 30  2  03 91         913   91.3 W       PDC
 32  4  00 02 5E 0B155147  155.147 kWh   YT
 36  2  00 58          88   88 Wh        YD

 DC CH3 (MPPT4)
 38  2  02 B5         693   69.3 V       UDC
 40  2  00 6F         111    1.11 A       IDC
 42  2  03 03         771   77.1 W       PDC
 44  4  00 02 50 0E151566  151.566 kWh   YT
 48  2  00 64         100   100 Wh       YD

 AC (3-phase)
 50  2  09 0D        2317  231.7 V       UAC_1N
 52  2  09 1C        2332  233.2 V       UAC_2N
 54  2  09 1F        2335  233.5 V       UAC_3N
 56  2  0F AB        4011  401.1 V       UAC_12
 58  2  0F D2        4050  405.0 V       UAC_23
 60  2  0F C1        4033  403.3 V       UAC_31
 62  2  13 88        5000   50.00 Hz     Freq
 64  2  00 00           0    0.0 VAR     Q
 66  2  08 CE        2254  225.4 W       PAC
 68  2  00 00           0   ???          unk
 70  2  00 E1         225   ???          unk
 72  2  00 22          34    0.34 A      IAC_1
 74  2  00 21          33    0.33 A      IAC_2
 76  2  00 1E          30    0.30 A      IAC_3
 78  2  03 E8        1000    1.000       PF

 Inverter
 80  2  00 78         120   12.0 °C      Temp
 82  2  00 03           3   3            EvtLog
 88  2  00 1A          26   ???          unk
 90  2  63 40       25408   CRC16
```

Full Decode
```
MIT-5000-8T RF CAPTURE — Complete Decode
=========================================

DC CH0 (MPPT1)        DC CH1 (MPPT2)
  UDC    33.7 V         UDC    35.4 V
  IDC    1.03 A         IDC    1.02 A
  PDC    34.9 W         PDC    36.8 W
  YT    80.368 kWh      YT    65.066 kWh
  YD       37 Wh        YD       35 Wh

DC CH2 (MPPT3)        DC CH3 (MPPT4)
  UDC    72.3 V         UDC    69.3 V
  IDC    1.26 A         IDC    1.11 A
  PDC    91.3 W         PDC    77.1 W
  YT   155.147 kWh      YT   151.566 kWh
  YD       88 Wh        YD      100 Wh

AC (3-phase)
  UAC 1N/2N/3N   231.7 / 233.2 / 233.5 V
  UAC 12/23/31   401.1 / 405.0 / 403.3 V
  Frequency      50.00 Hz
  PAC            225.4 W
  Q                0.0 VAR
  IAC 1/2/3      0.34 / 0.33 / 0.30 A
  PF             1.000

Inverter
  Temperature    12.0 °C
  Event Log      3

Unknown: offset 68 = 0, offset 70 = 225 (=PAC?)
```

matjhgc534z67umb, 2026-03-16:
Hey Guys, thanks for the effort you put into this request so far! I think a lot of people would love to see the support of the new Hoymiles Inverter Series implemented. So how is the current status? 

Thanks for your work!