tbnobody/OpenDTU #2774: OpenDTU master branch build fails

knxhm, 2025-07-24:
### What happened?

I tried to build OpenDTU with VScode and PlatformIO on Debian bookwork  the build completes with success mesage but it throws a lot of warnings a notes about failing macro expansions.
The image even would be flashed to the ESP32 board, but the device does not work.

I remember a few weeks earlier I was able to build the image on the same machine without problems.
To troubleshoot the issue, I did the same build operation on fresh install of Fedora42 but I run into the same problem.
And the same on a Windows11 and Ubuntu 24 machine as well. 

So I believe its not a build machine setup error but possibly some dependencies of the build environment with MACRO usage have changed.
The whole build log from WINDOWS is attached. 

### To Reproduce Bug

clone the OpenDTU master repo and start build of the generic-ESP32 with vscode and platformIO.

### Expected Behavior

The build should complete without MACRO expansion problems

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

3f01b8d

### What firmware variant (PIO Environment)?

generic_ESP32

### Relevant log/trace output

```Shell
build log file attached
```

### Anything else?

[opendtu_build_log.txt](https://github.com/user-attachments/files/21414931/opendtu_build_log.txt)

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

tbnobody, 2025-07-24:
The errors are normal behavior due to some issues in the arduino core used. They also occour in the CI build used on github. (Which are btw working very well). 
If the generated firmware blob is not working on your side it must be something different. Whats the output of the serial console?

knxhm, 2025-07-24:
there is no any output on serial console when i start it up. it looks like the whole serial subsystem is not working.

knxhm, 2025-07-24:
Seems I was wasn't patient enough. After I while I get on serial 

network: disabling search for AP 

and I can connect to the web interface.

Actually I compared the behaviour with AHOY DTU where the serial interface is much more chatty during startup.

This many MACRO expansion messages during build made me thinking the built image has a defect but if thats normal, then the issue is no issue. Thanks for support.