tbnobody/OpenDTU #2776: WR HM-400 ausgeschaltet

Burnett11, 2025-07-27:
### What happened?

Der WR-HM400 wird vom einem 5KW Speicher versorgt, alle HM WR funktionieren, nur der ein HM-400 nicht, ist ausgeschaltet und nicht erreichbar.

### To Reproduce Bug

Ich bin mir nicht sicher ob es am Smart Miro Solar liegt oder an der DTU Software v25.5.10 ?

### Expected Behavior

Ich werde aut die open DTU Version 25.2.3 wechseln

### Install Method

Pre-Compiled binary from GitHub releases

### What git-hash/version of OpenDTU?

	v4.4.7-dirty

### What firmware variant (PIO Environment)?

25.5.10

### Relevant log/trace output

```Shell

```

### Please confirm the following

- [x] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [x] I have already searched for relevant existing issues and discussions before opening this report.
- [x] I have updated the title field above with a concise description.
- [x] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported.

OutdoorRob, 2025-07-28:
Hallo @Burnett11 
du schreibst, dass der betroffene HM-400 **ausgeschaltet** ist - kannst du das bitte genauer spezifizieren?
Wenn du mit ausgeschaltet meinst, dass dieser keine Versorgungsspannung über die DC-Eingänge erhält und die LED weder grün noch rot blinkt, dann hat die OpenDTU keine Möglichkeit mit diesem zu kommunizieren (denn dazu benötigt dieser eine DC seitige Versorgung).

Burnett11, 2025-07-28:
Vielen Dank für Deine Antwort, ich habe die open DTU neu geflasht, jetzt sind alle 3 HM WR wieder da, Software v25.5.10, aber mein HMT 1800-6T kommuniziert nicht mit der DTU, weil das Funkmodul nicht konfigguriert ist. Habe die pin_mapping.json wieder hergestellt, aber trotzdem passiert nichts, hast Du einen Tip?Schön das Du geantwortet hast.Viele Grüße Jürgen--Gesendet mit der GMX Mail App

closed 2025-10-03 by github-actions[bot]