tbnobody/OpenDTU #2780: [Request] Set inverter to 0 when unreachable for X seconds

JT-web, 2025-08-02:
### Is your feature request related to a problem? Please describe.

Hey guys,

is it possible to add an option to set inverter to 0 when unreachable for X seconds? I use a Hoymiles HMS 1600 on a 5kwh LiFePo battery. When the battery reaches 5 percent, I use a relay to cut off the DC supply to the inverter. The problem is that from this point on, opendtu permanently displays the last feed-in value while the inverter is offline. My VenusOS installation accesses this value and thus permanently calculates the wrong feed-in. 

### Describe the solution you'd like

It would be great it there was a possibility to set the feed-in-value to zero in opendtu if the inverter loses connection.

spcqike, 2025-08-13:
you may change your behaviour. don't just disconnect the inverter, as the (hard) reconnect may damage it on the long run. (high inrush currents, if you dont have a soft start or rampup)

instead, stop the inverter and leave it connected. use the relay just as fallback, to not deep discharge your battery (in case your central smart home logic doesn't work as expected and doesn't stop your inverter in time)

like, stop inverter at 5%, disconnect at 2%.

anyway, don't only rely on SoC / percentage, as this might drift over time. most batteries don't recognize small loads properly. instead, use the voltage as threshold.

btw, you might have a look at the fork "openDTU-onBattery", as this can do this. you can setup your system in a way to protect your battery from to deep discharges, it will stop the production / the inverter at your threshold.

spcqike, 2025-08-13:
well, thats a valid point. i never realised this, as i don't use it :)
so thats way easier, i guess :)

However, he should not simply disconnect the battery, as this could impair the service life of the inverters. (depending on his battery, as some allready come with a soft start function)