tbnobody/OpenDTU #2812: Inverter Total Yield Day

bs6681, 2025-08-28:
### Describe the solution you'd like

please put "Inverter Total Yield Day"  in the same notification as the other date ,  such as "Inverter Total Yield Total"
.
And in Event log  please insert date, not just time

tbnobody, 2025-08-28:
> please put "Inverter Total Yield Day" in the same notification as the other date , such as "Inverter Total Yield Total"

Sorry, I don't know what you mean with this sentence.

> And in Event log please insert date, not just time

Date in event log is not possible as the date is not available in the data received from the inverter. Due to the fact that the eventlog is cleared every time the inverter turns offline (which happens at night), you can assume that all events for the current day

bs6681, 2025-08-28:
for example in English:  Inverter Total Yield Day=988.714 kWh but  Inverter Total Yield Day=36,088 Wh (point, comma)
in German    Inverter Total Yield Day=   988,716 kWh  but  Inverter Total Yield Day=36.090 Wh
Notation of numbers, decimal and thousand separators is indifferent, should be the same

tbnobody, 2025-08-28:
Hm no thats exactly right... you generated 988,716kWh in total and on that specific day 36kWh (= 36.090 Wh)

bs6681, 2025-08-28:
988,734 kWh  is in german  Notation of numbers, but 
36.108 Wh is  in english Notation of numbers .

see opendtu web display in german or in english  and you will see this different separators

bs6681, 2025-08-29:
it is OK .  Thank you 
I restarted the inverter, then the number is now small and the separator has become unimportant